olikraus/Ucglib_Arduino #1: Arduino Zero Slow

njsharkracer, 2016-04-03:
I have just tried v1.3.3 on my sparkfun samd21 mini breakout (zero clone) and your library works wonderfully.  In the FPS test i'm getting 7.1 for text, 10.5 for box, and 7.8 for gradient.

I have noticed from your previous speed tests that the DUE was also running much slower then the arduino uno.

My guess is the HW SPI speed for ARM controllers is running too slow.  I know that the samd21 has a max speed of 12Mhz on SPI.

Is there any code changes you can do for arm platform to speed up the spi speed?  I'm more then willing to test code changes for you.

olikraus, 2016-04-04:
I do not have access to a samd21 board, so there is no HW SPI support at the moment for it.

olikraus, 2017-12-12:
I will close this due to issue cleanup.

Please report here: https://github.com/olikraus/ucglib/issues

closed 2017-12-12 by olikraus