olikraus/Ucglib_Arduino #2: My ILI9163 shows the display shifted by 32 pixels from the connector side

beegee-tokyo, 2016-05-05:
I was testing your code with a ILI9163 driven display I got from [Elecrow](http://www.elecrow.com/wiki/index.php?title=1.44%27%27_128x_128_TFT_LCD_with_SPI_Interface)

Using your library, the display had an unused area (at the side where the connector is). Looking into your code, I saw
> Code is for 128x128 display which is shifted by 32 pixel within the controller RAM 
After removing the +/-32 from your ucg_dev_ic_ili9163.c file, the display works correct.
I did this by commenting out all lines with:
`ucg->arg.pixel.pos.y += 32;`
and
`ucg->arg.pixel.pos.y -= 32;`

Proposal:
add a #define or function to eliminate/define the offset to be compatible to displays without the offset. like:
```
#ifndef ILI9163NoShift
   ucg->arg.pixel.pos.y += 32;
#endif
#ifndef ILI9163NoShift
   ucg->arg.pixel.pos.y -= 32;
#endif
```
or 
```
   ucg->arg.pixel.pos.y += ucg->arg.pixel.shift;
   ucg->arg.pixel.pos.y -= ucg->arg.pixel.shift;
```
where ucg->arg.pixel.shift is by default 32 but can be set to any value in the 
`Ucglib_ILI9163_18x128x128_HWSPI ucg(/*cd=*/ 9 , /*cs=*/ 10, /*reset=*/ 8, /*shift=*/ 32);`
function call.

olikraus, 2016-05-05:
Interessting. At least my own display requires such shift.

beegee-tokyo, 2016-05-05:
There was the same issue reported some time ago: https://github.com/olikraus/ucglib/issues/60

olikraus, 2017-12-12:
I will close this due to issue cleanup.

Please update olikraus/ucglib#60

closed 2017-12-12 by olikraus