olikraus/Ucglib_Arduino #5: Compilation for stm32duino

karawin, 2017-03-27:
Hi,

I had added these line in Ucglib.h for stm32duino compilation:
 just after #include "clib/ucg.h" add:

 #if defined(__arm__)
  #define __NOP __asm__("nop\n\t")
 #endif

olikraus, 2017-03-27:
Can you give me some background, why this is needed?
Shouldn't it be better:
#if defined(arm)
#if !defined(__NOP)
#define __NOP asm("nop\n\t")
#endif
#endif

karawin, 2017-03-27:
The story is not clear. I prefer to close it now and study a bit more.
It seems that it is defined in 
Arduino\hardware\Arduino_STM32\STM32F4\cores\maple\libmaple\usbF4\VCP\core_cmInstr.h (3 hits)
	Line 46: #define __NOP                             __nop
	Line 269: __attribute__( ( always_inline ) ) static __INLINE void __NOP(void)

But in conflict with
#define __NOP __asm__("nop\n\t")

I have to check more
Sorry, I close it

olikraus, 2017-03-27:
Ok

karawin, 2017-04-02:
Back ...
I confirm that 👍 
in Ucglib.h for stm32duino compilation:
just after #include "clib/ucg.h" add:

#if defined(__arm__)
#undef __NOP
#define __NOP      __asm volatile ("nop")
#endif

must be added in order to port ucglib for stm32duino.
It is defined in standart library but not in the maple library used in STM32duino.
Work well.

olikraus, 2017-04-08:
ok, but isn't this a problem of stm32duino then?

karawin, 2017-04-08:
No, under stm32duino the WiringPinMode is an enum with more elements 
than the avr #define
#define INPUT 0x0
#define OUTPUT 0x1
#define INPUT_PULLUP 0x2,

so the pinmode procedure is typed:

void pinMode(uint8 pin, WiringPinMode mode);
/**
  * Specifies a GPIO pin behavior.
  * @see pinMode()
  */
typedef enum WiringPinMode {
     OUTPUT, /**< Basic digital output: when the pin is HIGH, the
                voltage is held at +3.3v (Vcc) and when it is LOW, it
                is pulled down to ground. */

     OUTPUT_OPEN_DRAIN, /**< In open drain mode, the pin indicates
                           "low" by accepting current flow to ground
                           and "high" by providing increased
                           impedance. An example use would be to
                           connect a pin to a bus line (which is pulled
                           up to a positive voltage by a separate
                           supply through a large resistor). When the
                           pin is high, not much current flows through
                           to ground and the line stays at positive
                           voltage; when the pin is low, the bus
                           "drains" to ground with a small amount of
                           current constantly flowing through the large
                           resistor from the external supply. In this
                           mode, no current is ever actually sourced
                           from the pin. */

     INPUT, /**< Basic digital input. The pin voltage is sampled; when
               it is closer to 3.3v (Vcc) the pin status is high, and
               when it is closer to 0v (ground) it is low. If no
               external circuit is pulling the pin voltage to high or
               low, it will tend to randomly oscillate and be very
               sensitive to noise (e.g., a breath of air across the pin
               might cause the state to flip). */

     INPUT_ANALOG, /**< This is a special mode for when the pin will be
                      used for analog (not digital) reads.  Enables ADC
                      conversion to be performed on the voltage at the
                      pin. */

     INPUT_PULLUP, /**< The state of the pin in this mode is reported
                      the same way as with INPUT, but the pin voltage
                      is gently "pulled up" towards +3.3v. This means
                      the state will be high unless an external device
                      is specifically pulling the pin down to ground,
                      in which case the "gentle" pull up will not
                      affect the state of the input. */

     INPUT_PULLDOWN, /**< The state of the pin in this mode is reported
                        the same way as with INPUT, but the pin voltage
                        is gently "pulled down" towards 0v. This means
                        the state will be low unless an external device
                        is specifically pulling the pin up to 3.3v, in
                        which case the "gentle" pull down will not
                        affect the state of the input. */

     INPUT_FLOATING, /**< Synonym for INPUT. */

     PWM, /**< This is a special mode for when the pin will be used for
             PWM output (a special case of digital output). */

     PWM_OPEN_DRAIN, /**< Like PWM, except that instead of alternating
                        cycles of LOW and HIGH, the voltage on the pin
                        consists of alternating cycles of LOW and
                        floating (disconnected). */
} WiringPinMode;

Anyway, i made my clone with the modified version and it works ok.
Thanks.

olikraus, 2017-04-09:
i am confused. What has your answer todo with the fact, that __NOP is undefined?

karawin, 2017-04-09:
You can. This message was not for you. Sorry sorry.

lacklustrlabs, 2017-11-17:
This is what i did to make use of this excellent library (thanks) with the current [stm32duino](http://www.stm32duino.com) repository:

I modified Ucglib.cpp to look something like this:
```
...
#include <SPI.h>

#include "Ucglib.h"

#if defined(__arm__)
#ifndef __NOP
#define __NOP __asm volatile ("nop")
#endif
#endif
...
```

Note that stm32duino does _not_ `#define arm` but it does `#define __arm__`

olikraus, 2017-12-12:
I will close this due to issue cleanup.

Please report here: https://github.com/olikraus/ucglib/issues

closed 2017-12-12 by olikraus