olikraus/Ucglib_Arduino #8: Ucglib_ILI9163_18x128x128_HWSPI slow on Teensy 3.2

Jorgen-VikingGod, 2017-04-25:
Hi @olikraus I tested some displays and I can see that the TFT displays are really "slow" on the Teensy 3.1/3.2.

Is there a way to use same technique as in the u8g2 lib - with clearBuffer() and sendBuffer() instead of writing out the data each time I call a method?

I really like the technique of the monochrome lib.

Or is it possible to "clear" only the text layer and let the background untouched?
I need to paint a nice "background" with gradient - and draw some boxes and texts on top.
But I do not want to repaint the background each time I change one text.

Maybe I can work here with clip areas?

Thx

Jorgen-VikingGod, 2017-05-06:
Nobody an idea how to speed up?

olikraus, 2017-05-08:
Hi

I probably need some time to investigate this further. Are you sure, that
HW SPI is used? Ucglib may fall back to SW SPI if the board is not
detected.
Actually I do not have time to work on ucglib :-(

Compared to u8g2, a TFT display would require 24 times more RAM. This would
be >200K for a 320x240 TFT for a complete page.
Most Arduino systems do not have that much RAM.

Oliver

olikraus, 2017-12-12:
I will close this due to issue cleanup.

Please report here: https://github.com/olikraus/ucglib/issues

closed 2017-12-12 by olikraus