olikraus/Ucglib_Arduino #13: Support for 80x160 ST7735 module

umarsear, 2018-08-05:
I am trying to use the library with an 80x160 0.96 display (IPS LCD), obviously this is currently an un-supported device in the library, however it does work with the obvious issue that the screen coordinates are off-set. 

What will it take to add support for this device.

olikraus, 2018-08-11:
currently I do not work on ucglib...

umarsear, 2018-08-11:
Any pointers on what I would need to change to add support. 

regards,

olikraus, 2018-08-12:
well, first you need to implement a new device structure:
https://github.com/olikraus/ucglib/blob/master/csrc/ucg_dev_tft_128x160_st7735.c#L85

make the new device known here: https://github.com/olikraus/ucglib/blob/master/csrc/ucg.h#L164

Create a new Arduino constructor, probably by cloning this constructor:
https://github.com/olikraus/ucglib/blob/master/cppsrc/Ucglib.h#L297

I think, thats it...

orhanyor, 2019-03-31:
> I am trying to use the library with an 80x160 0.96 display (IPS LCD), obviously this is currently an un-supported device in the library, however it does work with the obvious issue that the screen coordinates are off-set.
> 
> What will it take to add support for this device.

i was wondering if you managed to make it work? any luck?