olikraus/Ucglib_Arduino #16: Does drawXBMP function available in ucglib?

venkatesh223, 2019-04-25:
Does ucglib support drawxbmp like the U8g2 library?
If no can you help me with an alternative on how to display an image using ucglib arduino please.

olikraus, 2019-04-28:
I once created an example to show a pic in TGA format (https://en.wikipedia.org/wiki/Truevision_TGA): https://github.com/olikraus/ucglib/blob/master/sys/arduino/ShowTGA/ShowTGA.ino

venkatesh223, 2019-04-29:
This requires an SD Card to store tga file.
Anything with bitmap or with xbm file?

olikraus, 2019-04-29:
> This requires an SD Card to store tga file.

Obviously. But you can just use XBM instead. Will need a decoding of the XBM format, but this shouldn't be a big problem.

venkatesh223, 2019-07-16:
Yea will try.
Thanks for the support.

closed 2019-07-16 by venkatesh223