olikraus/Ucglib_Arduino #18: Ucglib tried on ST7789v need hep with that

venkatesh223, 2019-07-16:
Hello Olikraus,
I tried this library using a 2"inch 240X320 LCD driven by ST7789v the results were as shown in the image and video.
In the library i used ST7735 and connected this display.
To make it work proper will changing the resolution in all the respective files work? Or anything else needed to be done? Please help.

https://drive.google.com/open?id=1SYzbJIDt8TSCOI7K0Pri1pekVdHPPiMs

Will be waiting for your reply.
Regards 
Venkatesh
[image]

olikraus, 2019-07-16:
hmmm... difficult. I do not work on this project any more... and it is long time back...

venkatesh223, 2019-07-17:
Thank you for your reply.
So any suggestions on what can be done briefly.
I shall try it.

venkatesh223, 2019-07-19:
Hello there,
Huge thanks for the library,
as you know i was working with ST7789v driven display using your library.
its working finally. After many changes
Huge thanks again for the library.
[image]

closed 2019-07-19 by venkatesh223

olikraus, 2019-07-19:
wow

xmenxwk, 2021-11-09:
what changes have you made ?

charleslemaire0, 2022-12-26:
I am interested too