olikraus/Ucglib_Arduino #19: Lib does not work on Arduino Nano Every

PI-Beginner, 2019-07-19:
I have test this Library on a new Nano Every and it does not work.

Please can you update this great lib?

Thank you.

olikraus, 2019-07-19:
Why do you think that this lib doesn't work?

PI-Beginner, 2019-07-19:
I test it with your simple Example. First on Nano V3 (it run) then I changed the Microcontroller / Board to Nano Every and
 it produced many error message.

The boards are pin compatible but other controller.

olikraus, 2019-07-20:
> it produced many error message.

I do not own a "nano every". How can I fix anything without knowing the error messages? In fact I can not even start to analyse anything without known anyting about the problem.

Let me say it like this: I do not earn any money here. This is just a spare time project, which I wrote long time back. I would like to help you, but I do not have the time to ask you all the details (again: I will not be paied for this). It is your job to start the analysis why something is not working. If you want to approach me, you should provide ALL information you know about the problem and the setup (e.g. display type). Moreover you should as much as possible exclude any other problem.

Anyhow, I can not promise anything here. I have almost stopped working on this project...