olikraus/Ucglib_Arduino #21: Own font from bdf

hevet, 2019-11-12:
I have modified the bdf font. What command through bdfconv to generate a font from a bdf file to c format, and how do I correctly add it to the program code in Arduino IDE? Thanks

olikraus, 2019-11-13:
I think the commands are all listed here:
https://github.com/olikraus/ucglib/blob/master/tools/font/build/do_fonts_v2.sh

You can just copy the generated .c code to your .ino file.

hevet, 2019-11-13:
Thanks for the answer. Can you write me where to paste these commands in cmd or something else?

olikraus, 2019-11-14:
Can you write me where to paste these commands in cmd or something else?
You have to first build your own executable for bdfconv out of the source from here: https://github.com/olikraus/ucglib/tree/master/tools/font/bdfconv

With bdfconv available you can run the command, which will look like this:
```
bdfconv -b 0 -m '9632-9679>32' cu12.bdf -n ucg_font_cu12 -o cu12.c  
```

THe result will be a file called "cu12.c". You can just copy the content of that file directly in your .ino file and use the above name "ucg_font_cu12" (-n option) as argument for the setFont command.

Sergio-tix, 2022-07-23:
> You can just copy the content of that file directly in your .ino file and use the above name "ucg_font_cu12" (-n option) as argument for the setFont command.

What does it mean to add? I added the code and it doesn't work. Are there instructions on how to do this correctly?

[image]
And I couldn't find instructions for bdfconv.exe anywhere. Is there a command description?

olikraus, 2022-07-24:
You need to remove the include statement (i do not know from where it comes from).

Help on bdfconv.exe is printed if called without arguments in a cmd.exe shell.

Please also note, that work on ucglib has been halted.