olikraus/Ucglib_Arduino #27: у меня выдает ошибку usg

Sahsabox, 2023-07-22:
Arduino: 1.8.19 (Windows 10), Плата:"Arduino Uno"

C:\Users\sahsa\OneDrive\Рабочий стол\FPS\FPS.ino: In function 'void setup()':

FPS:146:3: error: 'ucg' was not declared in this scope

C:\Users\sahsa\OneDrive\Рабочий стол\FPS\FPS.ino: In function 'void draw_text()':

FPS:176:3: error: 'ucg' was not declared in this scope

C:\Users\sahsa\OneDrive\Рабочий стол\FPS\FPS.ino: In function 'void draw_box()':

FPS:189:3: error: 'ucg' was not declared in this scope

C:\Users\sahsa\OneDrive\Рабочий стол\FPS\FPS.ino: In function 'void draw_gradient_box()':

FPS:204:13: error: 'ucg' was not declared in this scope

FPS:216:3: error: 'ucg' was not declared in this scope

C:\Users\sahsa\OneDrive\Рабочий стол\FPS\FPS.ino: In function 'uint16_t measure(void (*)())':

FPS:225:3: error: 'ucg' was not declared in this scope

C:\Users\sahsa\OneDrive\Рабочий стол\FPS\FPS.ino: In function 'void show_result(const char*, uint16_t)':

FPS:276:3: error: 'ucg' was not declared in this scope

Несколько библиотек найдено для "Ucglib.h"

 Используется: C:\Users\sahsa\OneDrive\Документы\Arduino\libraries\Ucglib_Arduino-1.5.2

Не используется: C:\Users\sahsa\OneDrive\Документы\Arduino\libraries\Ucglib

exit status 1

'ucg' was not declared in this scope

Этот отчёт будет иметь больше информации с
включенной опцией Файл -> Настройки ->
"Показать подробный вывод во время компиляции"

olikraus, 2023-07-23:
Maybe you missed to uncomment the constructor

Sahsabox, 2023-07-23:
but how to fix it?

olikraus, 2023-07-23:
Uncomment the constructor line!